NebraLtd/hm-pktfwd #15: Packet Forwarder crashing out in AS923 region.

ryanteck, 2021-05-31:
The latest update of the miner (GA 29th of May) introduced the new regions of AS923_1, AS923_2, AS923_3 and AS923_4

As no warning was given of this change the packet forwarder doesn't have a matching index for these variables (Nor I think a frequency plan for AS923_3 & AS923_4).

These new plans need to be added, then updated on the index.

ryanteck, 2021-06-01:
Fixed in 2021.06.01.0

marvinmarnold, 2022-05-03:
This is still happening a lot on Sentry https://sentry.io/organizations/nebra/issues/2427112266/?project=5730185&statsPeriod=30d

shawaj, 2022-05-03:
What's the issue that's coming through in sentry @marvinmarnold ? Can't seem to log in

marvinmarnold, 2022-05-03:
[image]

shawaj, 2022-05-05:
That screenshot shows us915?

marvinmarnold, 2022-05-05:
It's the same codepath so Sentry groups them together. Whenever the region is unknown.

shawaj, 2022-05-07:
This seems like a miner that is not up to date. As key error for us915 / 923 implies outdated software (i.e. before we introduced those and it was just US915 without the region_ in front)

pritamghanghas, 2022-05-13:
sentry screenshots in case the sentry link is broken:
[image]
[image]
[image]

closed 2022-09-28 by KevinWassermann94