NebraLtd/hm-pktfwd #16: Rssi too strong compared to other hotspots

disk91, 2021-06-29:
At least for Europe, RSSI reported by some Nebra hotspot is really high compared to other nearest hotspot. As a consequence, the Nebra hotspot get Join Accept, ack and donwlink messages to process. This is applicable to hotspot used with outdoor antennas.
As an example, I've seen -80dBm > -90dBm over 13.8km. Free space path loss for 13.8km is 100dBm confirming the Nebra Rssi to be valid for 14dBm, but the problem it is really higher than the others.

Other hotspots are reporting for the same mesage:
- 7km / -118dBm (outdoor rak DiY)
- 8km / -115dBm (outdoor Rak hotspot)
- 8km / -107dBm (outdoor Rak hotspot)
Compared to Nebra
- 13,8km / -89dBm

As a consequence, since this Nebra is up, there is a real difficulty for devices in the city to join the network and receiving Ack.

shawaj, 2021-06-29:
Do you have further details of the hotspots in question so we can take a further look at them on the network?

disk91, 2021-06-29:
Take a look at Spare Lemon Urchin

shawaj, 2021-12-19:
> Take a look at Spare Lemon Urchin

For reference
https://explorer.helium.com/hotspots/11RRgLVeh7eigMp8wurCn4pCjfxhfk5iFz4CioP3MuYfmCapbv8/activity

@disk91 I don't see however that this is an issue with our concentrators, more what helium considers to be a valid witness in their acceptance curves

shawaj, 2021-12-19:
Interesting reading here also - people with other hardware with similar issues https://github.com/helium/miner/issues/618

rbrtio, 2022-01-04:
This may be impacted by the use, or lack of, a saw filter on the concentrator as they are generally introduce quite a bit of loss.

Another potential source of the issue is that some devices report rssi vs other that report rssis, which can be vastly different figures.

closed 2022-09-28 by KevinWassermann94