NebraLtd/hm-pktfwd #20: add check for spi bus config.txt line

shawaj, 2021-07-31:
if you do not enable the correct SPI bus, the lora concentrator fails to start with the error message...

```
ERROR: [main] failed to start the concentrator
```

however this is not really particularly helpful as an error message in this instance.

Perhaps there should be a check to see if the SPI bus defined in the [variant_definitions.py file](https://github.com/NebraLtd/helium-hardware-definitions/blob/master/variant_definitions.py) is actually enabled.

We can do this by searching for it in `/dev` using something like:
```
ls /dev | grep -i spi_bus
```

where spi_bus is the SPIBUS that is declared in variant_definitions.py

shawaj, 2021-07-31:
@vpetersson @robputt could you take a look at this shitty python code I have written and check if it looks ok 

https://github.com/NebraLtd/hm-pktfwd/blob/4216b49f982690228fc1e361c4ff3550a5195808/files/configurePktFwd.py#L22-L26

I am also not sure if it makes sense to keep trying the rest of the code if the SPI bus is not available? Maybe it makes more sense for it to just kill the container if this fails?

I guess this whole repo could do with a rewrite in any case....

vpetersson, 2021-07-31:
I don't think you can use pipes (`|`) like that when you shell out using `os.popen()`. Let's just rewrite this to native Python code instead, as there is really no reason to call on `os.popen()` at all when we can use native Python functions to interface with the file.

shawaj, 2021-07-31:
@vpetersson it works 😂 but not saying it's a good way of doing it.

I don't think /dev is a file but maybe we can use:

```
if os.path.exists('/dev/' + spi_bus)
    print("SPI bus Configured Correctly") 
else: 
     print("ERROR: SPI bus not found!") 
```

shawaj, 2021-07-31:
added in d29648e40d7d4c3d15389a425523ca45dd4c6350

let me know what you think @vpetersson

vpetersson, 2021-08-01:
`if os.path.exists('/dev/{}'.format(spi_bus)):` should work, but i don't have a device to easily test this.

shawaj, 2021-08-01:
@vpetersson will test it now

shawaj, 2021-08-01:
Added in https://github.com/NebraLtd/hm-pktfwd/pull/22/commits/d624b84cb937873ce44b1ba95f14bc87aaeb5492 and works fine

shawaj, 2021-08-03:
closing as this is added in now

closed 2021-08-03 by shawaj