NebraLtd/hm-pktfwd #21: str() function on a string?

shawaj, 2021-07-31:
When fixing some stuff in light-hotspot-software, I came across this:

https://github.com/NebraLtd/hm-pktfwd/blob/34e6a417039956404bbcab97ecf215e0efbd7872/files/configurePktFwd.py#L16-L19

The reset and spi_bus were back to front.

I have corrected it this:

https://github.com/NebraLtd/hm-pktfwd/blob/4216b49f982690228fc1e361c4ff3550a5195808/files/configurePktFwd.py#L16-L19

However due to my limited knowledge of code, I just wanted to get a sense check @vpetersson ... basically, I thought the str(spi_bus) was somewhat pointless as it is already a string? I am guessing the str() function was meant to be used for the reset variable as that is a number?

vpetersson, 2021-07-31:
Would be better to rewrite it as follows:

```
 spi_bus = variant_variables['SPIBUS'] 
 print("Hardware Variant {} detected".format(variant)) 
 print("RESET: {}.".format(reset_pin)) 
 print("SPI: {}".format(spi_bus))
```

This is usually how you write it in Python 3 style. The other way is more how you wrote it in Python 2.7.

shawaj, 2021-07-31:
corrected in d29648e40d7d4c3d15389a425523ca45dd4c6350

closed 2021-07-31 by shawaj