NebraLtd/hm-pktfwd #34: Packet forwarder failed to start concentrator

hades19, 2021-08-18:
Hi I have a problem with the concentrator is it possible that I missed something if I have the wrong variables in balena.cloud?
Or it's a problem in lora_templates_sx1302 / EU-global_conf.json.
My original system was probably damaged due to a poor Wi-Fi network. I have nebra indoor miner NEBHNT-IN1 for EU868. 
This is my balena.cloud variables
REGION_OVERRIDE	  not defined  EU868  packet-forwarder
VARIANT	  not defined   NEBHNT-IN1    diagnostics
VARIANT	  not defined   NEBHNT-IN1    gateway-config
VARIANT	  not defined   NEBHNT-IN1    packet-forwarder
Thank you for any help.

shawaj, 2021-08-18:
Is this a DIY hotspot?

hades19, 2021-08-18:
No, it is an original nebra indoor miner for the EU.
I had some problems with the system so I tried your software.
And I can't connect my miners to the nebra dashboard.
And this seems like a good solution.

shawaj, 2021-08-18:
You can't connect to the Nebra dashboard if you've manually installed this software and you also won't get automatic updates

hades19, 2021-08-20:
I understand. So this software is only for DIY.
It will not work properly with the original Miner?

vpetersson, 2021-08-20:
No, this is used for our own software. That said, if you reflash the unit at this time, you're on your own as this is not a supported use case at the moment.

closed 2021-08-20 by vpetersson