NebraLtd/hm-pktfwd #51: Issues with Australian miners

JohnWhiting, 2021-09-16:
Since update 2021.09.14.0 Nebra Miner users in Australia have been experiencing a lot of issues. Many of them have not received a successful witness / appearing offline since the update. This is the second time in two weeks the miners have been offline. 

Also, the Frequency is showing as 917.4 MHz instead of 915 MHz. 
[image]

vpetersson, 2021-09-16:
Looking into this now. 

What we have [here](https://github.com/NebraLtd/hm-pktfwd/blob/master/lora_templates_sx1302/AU-global_conf.json) is slightly different from [upstream](https://github.com/helium/sx1302_hal/blob/helium/hotspot/packet_forwarder/global_conf.json.sx1250.AU915).

The files are identical:

```
$ wget https://raw.githubusercontent.com/helium/sx1302_hal/helium/hotspot/packet_forwarder/global_conf.json.sx1250.AU915
> $ diff global_conf.json.sx1250.AU915 AU-global_conf.json
3c3
<     "spidev_path": "/dev/spidev0.0",
---
>     "spidev_path": "/dev/spidev1.2",
105,115c105,125
<     "gps_i2c_path": "/dev/i2c-1",
<     "gateway_ID": "AA555A0000000000",
<     "server_address": "localhost",
<     "serv_port_up": 1680,
<     "serv_port_down": 1680,
<     "keepalive_interval": 10,
<     "stat_interval": 30,
<     "push_timeout_ms": 100,
<     "forward_crc_valid": true,
<     "forward_crc_error": false,
<     "forward_crc_disabled": false
---
>       "gateway_ID": "AA555A0000000000",
>
>       "server_address": "helium-miner",
>       "serv_port_up": 1680,
>       "serv_port_down": 1680,
>
>       "keepalive_interval": 10,
>       "stat_interval": 30,
>       "push_timeout_ms": 100,
>
>       "forward_crc_valid": true,
>       "forward_crc_error": false,
>       "forward_crc_disabled": false
>   },
>
>   "debug_conf": {
>       "ref_payload":[
>           {"id": "0xCAFE1234"},
>           {"id": "0xCAFE2345"}
>       ],
>       "log_file": "loragw_hal.log"
```

That said, I don't see any difference with regards to frequency.

vpetersson, 2021-09-16:
Relevant screenshot from NebraLtd/hm-miner#26

[image]

shawaj, 2021-09-16:
We don't use the 918.2mhz channel anymore so this is not possible unless the Hotspot in question hasn't got the latest update

shawaj, 2021-09-16:
@vpetersson just FYI we don't use the sx1302 configs we use sx1301

shawaj, 2021-09-16:
https://github.com/NebraLtd/hm-pktfwd/blob/master/lora_templates_sx1301/AU-global_conf.json

vpetersson, 2021-09-16:
> https://github.com/NebraLtd/hm-pktfwd/blob/master/lora_templates_sx1301/AU-global_conf.json

@shawaj where does this file come from? I can't find it in any of the Helium repos.

jaytheblader, 2021-09-16:
> Since update 2021.09.14.0 Nebra Miner users in Australia have been experiencing a lot of issues. Many of them have not received a successful witness / appearing offline since the update. This is the [...]

This is probably the best example where this issue is occurring as it shows all the Nebra gateways witness events are invalid due to incorrect channel yet the Raks are fine with the Beacon.

https://explorer.helium.com/txns/Rs7RFFlkMJw9Bit-c7OpgMp7dXnx3AD_AMQ_B-KzQac

cryptowinteam, 2021-09-17:
Hey guys, I am sure you are working on many issues but wanted to see if there is an ETA of a rollout of a fix for this issue.  Given the large % of Nebra in Australia this is having a major impact.  If you have an estimate I will gladly share this via of social media group.

vpetersson, 2021-09-17:
@cryptowinteam the fix should have been rolled out. Long story short, it was a bug in the upstream code for Australia that we deployed, which was later fixed (which we've also rolled out). If you're still having issues, get in touch with support@nebra.com and we should be able to resolve it.

shawaj, 2021-09-17:
> > https://github.com/NebraLtd/hm-pktfwd/blob/master/lora_templates_sx1301/AU-global_conf.json
> 
> @shawaj where does this file come from? I can't find it in any of the Helium repos.

helium don't use the sx1301 in their miners so they don't have configs for it...

shawaj, 2021-09-17:
> This is probably the best example where this issue is occurring as it shows all the Nebra gateways witness events are invalid due to incorrect channel yet the Raks are fine with the Beacon. [...]

@jaytheblader that is showing a valid witness by the looks of it? 917.4 is a valid freq for australia. 915mhz is not. See here - https://github.com/helium/miner/blob/229e0bff969703ae52c989c319af24f48556931f/config/sys.config#L115

shawaj, 2021-09-21:
Think this can be closed. I've checked the configs many times and can't find any issues with our configs

jaytheblader, 2021-09-21:
@shawaj i agree, whatever was causing the issue is now resolved.

shawaj, 2021-09-22:
@jaytheblader thanks for the update.

Please keep us posted though, and if you notice anything else strange then fell free to open a new issue here in the future.

Glad it's all working better now 👍🙂

closed 2021-09-22 by shawaj