NebraLtd/hm-pktfwd #53: AS923_1 asserted cause lora operational false

hollejacklin, 2021-10-04:
I got two new nebra indoor miner 2021/9/27. I power on the first one, and local diagnostic showed all ok. However the local diagnostic showed lora operational error after the region asserted to AS923_1.
I power on the another one. The local diagnostic showed all ok in the beginning, and then the lora operational became false, after AS923_1 was asserted to region.
Please check this issue. my ticket 282366

shawaj, 2021-10-05:
@hollejacklin thanks for reporting this. A fix is coming in #54

closed 2021-10-05 by shawaj

shawaj, 2021-10-05:
Testing the fix now https://github.com/NebraLtd/helium-miner-software/commit/561767f8d3c8ef27ac61b214e7ce1efbfa81823a

shawaj, 2021-10-05:
Config working now - just going to give it a few hours before merging to mainnet

shawaj, 2021-10-05:
PR here https://github.com/NebraLtd/helium-miner-software/pull/158

shawaj, 2021-10-05:
@hollejacklin should be fixed now 👍

hollejacklin, 2021-10-05:
Yes it work. well done

PR #54 (2021-10-05): Remove listen before talk
Remove listen before talk

Closes: #53

**Why**

**How**

**References**
commit 0cf118d1 (2021-10-05): Remove listen before talk

Remove listen before talk

Closes: #53