NebraLtd/hm-pktfwd #70: Sx1302 not working

shawaj, 2021-11-15:
Sx1302 / sx1303 not working with current packet forwarder. Need to investigate why

shawaj, 2021-11-15:
Manually running chip_id on SX1301 based device:
```
root@0f1e195f0f9c:/opt/sx1302# ./chip_id 
CONCENTRATOR_RESET_PIN parameter not passed in, using value from the environment (val=)
./reset_lgw.sh: 33: echo: echo: I/O error
./reset_lgw.sh: 36: ./reset_lgw.sh: cannot create /sys/class/gpio/gpio/direction: Directory nonexistent
CoreCell reset through GPIO...
Opening SPI communication interface
Note: chip version is 0x00 (v0.0)
ERROR: Failed to set SX1250_0 in STANDBY_RC mode
ERROR: failed to setup radio 0
ERROR: failed to start the gateway
```
This correctly gives a 1 exit code

On SX1302 based:
```
root@b020abb3ab4e:/opt# cd sx1302
root@b020abb3ab4e:/opt/sx1302# ./chip_id -d /dev/spidev1.2
CONCENTRATOR_RESET_PIN parameter not passed in, using value from the environment (val=)
./reset_lgw.sh: 33: echo: echo: I/O error
./reset_lgw.sh: 36: ./reset_lgw.sh: cannot create /sys/class/gpio/gpio/direction: Directory nonexistent
CoreCell reset through GPIO...
Opening SPI communication interface
Note: chip version is 0x10 (v1.0)
INFO: using legacy timestamp
ARB: dual demodulation disabled for all SF

INFO: concentrator chip model ID: 0x00

INFO: concentrator EUI: 0x0016c001ff1a08f0

Closing SPI communication interface
CONCENTRATOR_RESET_PIN parameter not passed in, using value from the environment (val=)
CoreCell reset through GPIO...
```
This correctly gives a 0 exit code

However when running automatically on an sx1302 device, it is not recognised as being sx1302 based.

Issue, I guess, is something to do with the check:
https://github.com/NebraLtd/hm-pktfwd/blob/3f7c41eedfcbda414faec4d2cc13e149d7ca704e/pktfwd/utils.py#L62-L76

Or possibly how we use it later.

These devices on balena are running sx1302
https://dashboard.balena-cloud.com/devices/28c44310bfc8ffc6098bce2b3696336e/summary
https://dashboard.balena-cloud.com/devices/7f17117b0ff743debb90055bf21ceaa7/summary

shawaj, 2021-11-15:
@marvinmarnold I guess from [python docs](https://docs.python.org/3/library/subprocess.html) it needs to be without the stdout like this:

```
subprocess.run(util_chip_id_cmd, capture_output=True, 
                        text=True, check=True)
```
as we aren't using stdout for anything currently?

maybe in future we will want to use the output to determine if sx1302 or sx1303 but not sure if that maybe is best in a different function? not sure what you think?

shawaj, 2021-11-15:
created a PR in #71 - unless you think there is a better way to do this?

shawaj, 2021-11-15:
Ok, that didn't seem to work anyway! 🤦

shawaj, 2021-11-15:
On second thought, I guess this is the issue:
```
./reset_lgw.sh: 33: echo: echo: I/O error
./reset_lgw.sh: 36: ./reset_lgw.sh: cannot create /sys/class/gpio/gpio/direction: Directory nonexistent
```
As it's probably throwing an exception

shawaj, 2021-11-15:
Weirdly, it also seems to try to reset twice....

```
 packet-forwarder          SX1302_LORA_PKT_FWD_FILEPATH=/opt/sx1302/lora_pkt_fwd
 packet-forwarder          SX1301_LORA_PKT_FWD_DIR=/opt/sx1301
 packet-forwarder  
 packet-forwarder  2021-11-15 16:42:32,945 - [DEBUG] - pktfwd.pktfwd_app - (pktfwd_app.py).set_variant_attributes -- /opt/pktfwd/pktfwd_app.py:(91) - Variant COMP-RAKHM set with reset_pin 17 and spi_bus spidev0.0
 packet-forwarder  2021-11-15 16:42:32,946 - [DEBUG] - pktfwd.pktfwd_app - (pktfwd_app.py).start -- /opt/pktfwd/pktfwd_app.py:(39) - STARTING PKTFWD
 packet-forwarder  2021-11-15 16:42:32,948 - [DEBUG] - hm_pyhelper.miner_param - (miner_param.py).await_spi_available -- /opt/pktfwd-dependencies/hm_pyhelper/miner_param.py:(217) - SPI bus spidev0.0 Configured Correctly
 packet-forwarder  2021-11-15 16:42:32,950 - [DEBUG] - pktfwd.pktfwd_app - (pktfwd_app.py).prepare_to_start -- /opt/pktfwd/pktfwd_app.py:(73) - Region set to EU868
 packet-forwarder  2021-11-15 16:42:32,950 - [DEBUG] - pktfwd.utils - (utils.py).await_system_ready -- /opt/pktfwd/utils.py:(49) - Waiting 5 seconds for systems to be ready
 packet-forwarder  2021-11-15 16:42:37,956 - [DEBUG] - pktfwd.utils - (utils.py).await_system_ready -- /opt/pktfwd/utils.py:(51) - System now ready
 packet-forwarder  2021-11-15 16:42:37,957 - [DEBUG] - pktfwd.pktfwd_app - (pktfwd_app.py).prepare_to_start -- /opt/pktfwd/pktfwd_app.py:(76) - Finished preparing pktfwd
 packet-forwarder  2021-11-15 16:42:37,971 - [DEBUG] - pktfwd.utils - (utils.py).update_global_conf -- /opt/pktfwd/utils.py:(99) - No chip EUI. Assume SX1301
 packet-forwarder  2021-11-15 16:42:37,972 - [DEBUG] - pktfwd.utils - (utils.py).replace_sx1301_global_conf_with_regional -- /opt/pktfwd/utils.py:(116) - Copying SX1301 conf from /opt/pktfwd/config/lora_templates_sx1301/EU-global_conf.json to /opt/global_conf.json
 packet-forwarder  CONCENTRATOR_RESET_PIN parameter not passed in, using value from the environment (val=17)
 packet-forwarder  CoreCell reset through GPIO17...
 packet-forwarder  CONCENTRATOR_RESET_PIN parameter not passed in, using value from the environment (val=17)
 packet-forwarder  CoreCell reset through GPIO17...
```

marvinmarnold, 2021-11-15:
> Weirdly, it also seems to try to reset twice....

I think this is because chip_id [calls reset](https://github.com/NebraLtd/sx1302_hal/blob/master/util_chip_id/src/chip_id.c#L155) and then immediately after we explicitly do another reset before starting the concentrator. Its a little weird that the `replace_sx1301_global_conf_with_regional` logline doesn't show up in between the two resets, but given there's a mix of logs coming from python vs C, not too surprising.

marvinmarnold, 2021-11-16:
#72 is only a partial fix. With that change, the correct concentrator version is now detected. But it still fails to start:

```
2021-11-16 02:54:25,361 - [DEBUG] - pktfwd.utils - (utils.py).await_system_ready -- /opt/pktfwd/utils.py:(49) - Waiting 5 seconds for systems to be ready
2021-11-16 02:54:30,368 - [DEBUG] - pktfwd.utils - (utils.py).await_system_ready -- /opt/pktfwd/utils.py:(51) - System now ready
2021-11-16 02:54:30,370 - [DEBUG] - pktfwd.pktfwd_app - (pktfwd_app.py).prepare_to_start -- /opt/pktfwd/pktfwd_app.py:(76) - Finished preparing pktfwd
2021-11-16 02:54:32,601 - [DEBUG] - pktfwd.utils - (utils.py).is_concentrator_sx1302 -- /opt/pktfwd/utils.py:(72) - SX1302 / SX1303 Detected
2021-11-16 02:54:37,626 - [ERROR] - pktfwd.utils - (utils.py).retry_start_concentrator -- /opt/pktfwd/utils.py:(185) - lora_pkt_fwd stopped with code=1. 
```

shawaj, 2021-11-16:
Hmmm strange.

shawaj, 2021-11-16:
@marvinmarnold just had a thought. What happens if you put the rest script both in the old location and the new location?

It used to work, so I'm wondering if the packet forwarder is expecting it somewhere else?

shawaj, 2021-11-16:
Actually it's showing a message saying it can't find global_conf.json so maybe that's in the wrong place:
https://github.com/NebraLtd/hm-pktfwd/blob/3f7c41eedfcbda414faec4d2cc13e149d7ca704e/pktfwd/utils.py#L128-L130

Think this should be sx1302 directory or root dir?

shawaj, 2021-11-22:
testing in https://github.com/NebraLtd/helium-miner-software/pull/264 and this issue will be closed when pushed to production

marvinmarnold, 2021-11-29:
This is in production now. Closing.

closed 2021-11-29 by marvinmarnold

shawaj, 2022-02-18:
@Benixpearlin probably an incorrect reset pin script or configuration file