NebraLtd/hm-pktfwd #98: X86_64 compatibility

KevinWassermann94, 2022-07-05:
We need the packet forwarder to be compatible with x86_64 so that we can start the FCC certification

posterzh, 2022-07-27:
[Lora compliance testing](https://github.com/NebraLtd/compliance-testing/blob/master/rockpi/4b/lora/start-lora-test.sh) uses [sx1302_hal](https://github.com/NebraLtd/sx1302_hal) and hardcoded SPI device. 
Since we need to use USB concentrator in this x86 board, we need to use USB device in `sx1302_hal`. It needs to update USB firmware of concentrator.

posterzh, 2022-07-28:
@KevinWassermann94's x86 device on Balena recognized the RAK USB concentrator.
[image]

However, my x86 on Balena doesn't recognize the Nebra USB concentrator.
[image]

I expect to find `/dev/ttyACMx` but I couldn't find on my x86.
It looks like Nebra USB concentrator is not programmed with USB firmware.

Since I couldn't find `/dev/ttyACMx`, I couldn't program USB firmware for [sx1302_hal](https://github.com/Lora-net/sx1302_hal/tree/master/util_boot).
[image]

@MuratUrsavas Is there any way to flash STM32 firmware in Nebra concentrator so that it can be changed from 
`Bus 002 Device 002: ID 0483:df11 STMicroelectronics STM32  BOOTLOADER` 
to 
`Bus 002 Device 002: ID 0483:5740 STMicroelectronics STM32 Virtual ComPort` ?

KevinWassermann94, 2022-07-28:
@posterzh I will put the Nebra USB concentrator in mine. Then let's test again

posterzh, 2022-07-28:
[image]

Yours is same with mine.

x86 board and Balena HostOS recognized the USB devices correctly.
But the Nebra USB concentrator is not programmed its STM32 with USB firmware in the HPT.

posterzh, 2022-08-03:
On Kevin's x86 device, RAK USB concentrator is plugged and Ubuntu amd64 container named `qctesting` is running.
Inside the Ubuntu container:

#### 1. Built all the executables required by [Lora QC test script](https://github.com/NebraLtd/compliance-testing/blob/master/rockpi/4b/lora/start-lora-test.sh). 
  - QC script uses `test_loragw_hal_tx`, `test_loragw_hal_rx` built from https://github.com/NebraLtd/sx1302_hal.
  [image]

  - QC script uses `util_tx_test`, `util_tx_continuous` built from https://github.com/NebraLtd/compliance-testing/tree/master/rockpi/4b/lora

-   [image]

  

-   [image]

#### 2. USB concentrator is recognized as `/dev/ttyACM0` in the Balena Host OS and inside the Ubuntu container.

- **Host OS**
  [image]

- **Ubuntu Container**
  [image]

posterzh, 2022-08-10:
`helium-miner` container and `packet-forwarder` are running on x86 balena device. Also the global packet-forwader configuration for a region is set to use USB concentrator, for ex, `EU-global_conf.json`:
```
{
  "SX130x_conf": {
      "com_type": "USB",
      "com_path": "/dev/ttyACM0",
      "lorawan_public": true,
      ...
}
```

@MuratUrsavas, however the packet-forwader failed to communicate with the USB concentrator.
```
2022-08-10 12:37:56,017 - [DEBUG] - __main__ - (__main__.py).validate_env -- /opt/pktfwd/__main__.py:(63) - Starting with the following ENV:
        VARIANT=x86
        REGION_OVERRIDE=EU868
        REGION_FILEPATH=/var/pktfwd/region
        SX1301_REGION_CONFIGS_DIR=/opt/pktfwd/config/lora_templates_sx1301
        SX1302_REGION_CONFIGS_DIR=/opt/pktfwd/config/lora_templates_sx1302
        SENTRY_DSN=False
        BALENA_ID=de663284c63560c64ed9a6e3c9ae0ec1
        BALENA_APP=5G-Testing
        DIAGNOSTICS_FILEPATH=/var/pktfwd/diagnostics
        AWAIT_SYSTEM_SLEEP_SECONDS=5
        RESET_LGW_FILEPATH=/opt/reset_lgw.sh
        UTIL_CHIP_ID_FILEPATH=/opt/sx1302/chip_id
        ROOT_DIR=/opt
        SX1302_LORA_PKT_FWD_FILEPATH=/opt/sx1302/lora_pkt_fwd
        SX1301_LORA_PKT_FWD_DIR=/opt/sx1301

2022-08-10 12:37:56,018 - [DEBUG] - pktfwd.pktfwd_app - (pktfwd_app.py).set_variant_attributes -- /opt/pktfwd/pktfwd_app.py:(92) - Variant x86 set with reset_pin  and spi_bus 
2022-08-10 12:37:56,018 - [DEBUG] - pktfwd.pktfwd_app - (pktfwd_app.py).start -- /opt/pktfwd/pktfwd_app.py:(39) - STARTING PKTFWD
2022-08-10 12:37:56,019 - [DEBUG] - pktfwd.pktfwd_app - (pktfwd_app.py).prepare_to_start -- /opt/pktfwd/pktfwd_app.py:(74) - Region set to EU868
2022-08-10 12:37:56,019 - [DEBUG] - pktfwd.utils - (utils.py).await_system_ready -- /opt/pktfwd/utils.py:(57) - Waiting 5 seconds for systems to be ready
2022-08-10 12:38:01,024 - [DEBUG] - pktfwd.utils - (utils.py).await_system_ready -- /opt/pktfwd/utils.py:(59) - System now ready
2022-08-10 12:38:01,025 - [DEBUG] - pktfwd.pktfwd_app - (pktfwd_app.py).prepare_to_start -- /opt/pktfwd/pktfwd_app.py:(77) - Finished preparing pktfwd
2022-08-10 12:38:01,025 - [DEBUG] - pktfwd.utils - (utils.py).replace_sx1302_global_conf_with_regional -- /opt/pktfwd/utils.py:(171) - Injecting SPI bus  into global conf
2022-08-10 12:38:01,026 - [DEBUG] - pktfwd.utils - (utils.py).replace_sx1302_global_conf_with_regional -- /opt/pktfwd/utils.py:(177) - Saving SX1302 global conf from /opt/pktfwd/config/lora_templates_sx1302/EU-global_conf.json to /opt/global_conf.json with spi bus 
2022-08-10 12:38:01,028 - [DEBUG] - pktfwd.utils - (utils.py).replace_sx1302_global_conf_with_regional -- /opt/pktfwd/utils.py:(183) - Copying SX1302 local conf from /opt/pktfwd/config/lora_templates_sx1302/local_conf.json to /opt/local_conf.json
*** Packet Forwarder ***
Version: 2.1.0
*** SX1302 HAL library version info ***
Version: 2.1.0;
***
INFO: Little endian host
INFO: found configuration file global_conf.json, parsing it
INFO: global_conf.json does contain a JSON object named SX130x_conf, parsing SX1302 parameters
INFO: com_type USB, com_path /dev/, lorawan_public 1, clksrc 0, full_duplex 0
INFO: antenna_gain 0 dBi
INFO: Configuring legacy timestamp
INFO: no configuration for SX1261
INFO: Configuring Tx Gain LUT for rf_chain 0 with 16 indexes for sx1250
INFO: radio 0 enabled (type SX1250), center frequency 867500000, RSSI offset -215.399994, tx enabled 1, single input mode 0
INFO: radio 1 enabled (type SX1250), center frequency 868500000, RSSI offset -215.399994, tx enabled 0, single input mode 0
INFO: no configuration for LoRa multi-SF spreading factors enabling
INFO: Lora multi-SF channel 0>  radio 1, IF -400000 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora multi-SF channel 1>  radio 1, IF -200000 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora multi-SF channel 2>  radio 1, IF 0 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora multi-SF channel 3>  radio 0, IF -400000 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora multi-SF channel 4>  radio 0, IF -200000 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora multi-SF channel 5>  radio 0, IF 0 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora multi-SF channel 6>  radio 0, IF 200000 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora multi-SF channel 7>  radio 0, IF 400000 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora std channel> radio 1, IF -200000 Hz, 250000 Hz bw, SF 7, Explicit header
INFO: FSK channel 8 disabled
INFO: global_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
INFO: gateway MAC address is configured to AA555A0000000000
INFO: server hostname or IP address is configured to "helium-miner"
INFO: upstream port is configured to "1680"
INFO: downstream port is configured to "1680"
[... 5 lines cut ...]
INFO: packets received with no CRC will NOT be forwarded
INFO: global_conf.json does contain a JSON object named debug_conf, parsing debug parameters
INFO: got 2 debug reference payload
INFO: reference payload ID 0 is 0xCAFE1234
INFO: reference payload ID 1 is 0xCAFE2345
INFO: setting debug log file name to loragw_hal.log
INFO: found configuration file local_conf.json, parsing it
INFO: local_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
INFO: gateway MAC address is configured to AA555A0000000000
INFO: packets received with a valid CRC will be forwarded
INFO: packets received with a CRC error will NOT be forwarded
INFO: packets received with no CRC will NOT be forwarded
INFO: Reference latitude is configured to 0.000000 deg
[... cut ...]
2022-08-10 12:38:06,036 - [WARNING] - pktfwd.pktfwd_app - (pktfwd_app.py).start -- /opt/pktfwd/pktfwd_app.py:(60) - Shutting down concentrator.
2022-08-10 12:38:06,036 - [DEBUG] - pktfwd.pktfwd_app - (pktfwd_app.py).stop -- /opt/pktfwd/pktfwd_app.py:(80) - STOPPING PKTFWD
2022-08-10 12:38:06,037 - [INFO] - __main__ - (__main__.py).start -- /opt/pktfwd/__main__.py:(106) - Stopping and cleaning up pktfwd
2022-08-10 12:38:06,037 - [DEBUG] - pktfwd.pktfwd_app - (pktfwd_app.py).stop -- /opt/pktfwd/pktfwd_app.py:(80) - STOPPING PKTFWD
```

posterzh, 2022-08-10:
Inside the `packet-forwarder` container, USB concentrator is recognized, but it looks like USB-serial bridge is not working.
[image]

posterzh, 2022-08-11:
`helium-miner` and `packet-forwarder` services are working.

`packet-forwarder` logs:
```
2022-08-11 17:29:22,094 - [DEBUG] - __main__ - (__main__.py).validate_env -- /opt/pktfwd/__main__.py:(63) - Starting with the following ENV:
        VARIANT=x86
        REGION_OVERRIDE=EU868
        REGION_FILEPATH=/var/pktfwd/region
        SX1301_REGION_CONFIGS_DIR=/opt/pktfwd/config/lora_templates_sx1301
        SX1302_REGION_CONFIGS_DIR=/opt/pktfwd/config/lora_templates_sx1302
        SENTRY_DSN=False
        BALENA_ID=de663284c63560c64ed9a6e3c9ae0ec1
        BALENA_APP=5G-Testing
        DIAGNOSTICS_FILEPATH=/var/pktfwd/diagnostics
        AWAIT_SYSTEM_SLEEP_SECONDS=5
        RESET_LGW_FILEPATH=/opt/reset_lgw.sh
        UTIL_CHIP_ID_FILEPATH=/opt/sx1302/chip_id
        ROOT_DIR=/opt
        SX1302_LORA_PKT_FWD_FILEPATH=/opt/sx1302/lora_pkt_fwd
        SX1301_LORA_PKT_FWD_DIR=/opt/sx1301

2022-08-11 17:29:22,095 - [DEBUG] - pktfwd.pktfwd_app - (pktfwd_app.py).set_variant_attributes -- /opt/pktfwd/pktfwd_app.py:(92) - Variant x86 set with reset_pin  and spi_bus 
2022-08-11 17:29:22,095 - [DEBUG] - pktfwd.pktfwd_app - (pktfwd_app.py).start -- /opt/pktfwd/pktfwd_app.py:(39) - STARTING PKTFWD
2022-08-11 17:29:22,096 - [DEBUG] - pktfwd.pktfwd_app - (pktfwd_app.py).prepare_to_start -- /opt/pktfwd/pktfwd_app.py:(74) - Region set to EU868
2022-08-11 17:29:22,097 - [DEBUG] - pktfwd.utils - (utils.py).await_system_ready -- /opt/pktfwd/utils.py:(57) - Waiting 5 seconds for systems to be ready
2022-08-11 17:29:27,101 - [DEBUG] - pktfwd.utils - (utils.py).await_system_ready -- /opt/pktfwd/utils.py:(59) - System now ready
2022-08-11 17:29:27,102 - [DEBUG] - pktfwd.pktfwd_app - (pktfwd_app.py).prepare_to_start -- /opt/pktfwd/pktfwd_app.py:(77) - Finished preparing pktfwd
2022-08-11 17:29:27,103 - [DEBUG] - pktfwd.utils - (utils.py).replace_sx1302_global_conf_with_regional -- /opt/pktfwd/utils.py:(171) - Injecting SPI bus  into global conf
2022-08-11 17:29:27,103 - [DEBUG] - pktfwd.utils - (utils.py).replace_sx1302_global_conf_with_regional -- /opt/pktfwd/utils.py:(177) - Saving SX1302 global conf from /opt/pktfwd/config/lora_templates_sx1302/EU-global_conf.json to /opt/global_conf.json with spi bus 
2022-08-11 17:29:27,106 - [DEBUG] - pktfwd.utils - (utils.py).replace_sx1302_global_conf_with_regional -- /opt/pktfwd/utils.py:(183) - Copying SX1302 local conf from /opt/pktfwd/config/lora_templates_sx1302/local_conf.json to /opt/local_conf.json
*** Packet Forwarder ***
Version: 2.1.0
*** SX1302 HAL library version info ***
Version: 2.1.0;
***
INFO: Little endian host
INFO: found configuration file global_conf.json, parsing it
INFO: global_conf.json does contain a JSON object named SX130x_conf, parsing SX1302 parameters
INFO: com_type USB, com_path /dev/ttyACM0, lorawan_public 1, clksrc 0, full_duplex 0
INFO: antenna_gain 0 dBi
INFO: Configuring legacy timestamp
INFO: no configuration for SX1261
INFO: Configuring Tx Gain LUT for rf_chain 0 with 16 indexes for sx1250
INFO: radio 0 enabled (type SX1250), center frequency 867500000, RSSI offset -215.399994, tx enabled 1, single input mode 0
INFO: radio 1 enabled (type SX1250), center frequency 868500000, RSSI offset -215.399994, tx enabled 0, single input mode 0
INFO: no configuration for LoRa multi-SF spreading factors enabling
INFO: Lora multi-SF channel 0>  radio 1, IF -400000 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora multi-SF channel 1>  radio 1, IF -200000 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora multi-SF channel 2>  radio 1, IF 0 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora multi-SF channel 3>  radio 0, IF -400000 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora multi-SF channel 4>  radio 0, IF -200000 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora multi-SF channel 5>  radio 0, IF 0 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora multi-SF channel 6>  radio 0, IF 200000 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora multi-SF channel 7>  radio 0, IF 400000 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora std channel> radio 1, IF -200000 Hz, 250000 Hz bw, SF 7, Explicit header
INFO: FSK channel 8 disabled
INFO: global_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
INFO: gateway MAC address is configured to AA555A0000000000
INFO: server hostname or IP address is configured to "helium-miner"
INFO: upstream port is configured to "1680"
INFO: downstream port is configured to "1680"
INFO: downstream keep-alive interval is configured to 10 seconds
INFO: statistics display interval is configured to 30 seconds
INFO: upstream PUSH_DATA time-out is configured to 100 ms
INFO: packets received with a valid CRC will be forwarded
INFO: packets received with a CRC error will NOT be forwarded
INFO: packets received with no CRC will NOT be forwarded
INFO: global_conf.json does contain a JSON object named debug_conf, parsing debug parameters
INFO: got 2 debug reference payload
INFO: reference payload ID 0 is 0xCAFE1234
[... 58 lines cut ...]
# CRC_OK: 0.00%, CRC_FAIL: 0.00%, NO_CRC: 0.00%
# RF packets forwarded: 0 (0 bytes)
# PUSH_DATA datagrams sent: 1 (162 bytes)
# PUSH_DATA acknowledged: 0.00%
### [DOWNSTREAM] ###
# PULL_DATA sent: 3 (0.00% acknowledged)
# PULL_RESP(onse) datagrams received: 0 (0 bytes)
# RF packets sent to concentrator: 0 (0 bytes)
# TX errors: 0
### SX1302 Status ###
# SX1302 counter (INST): 61281034
# SX1302 counter (PPS):  0
# BEACON queued: 0
# BEACON sent so far: 0
# BEACON rejected: 0
### [JIT] ###
src/jitqueue.c:440:jit_print_queue(): INFO: [jit] queue is empty
#--------
src/jitqueue.c:440:jit_print_queue(): INFO: [jit] queue is empty
### [GPS] ###
# GPS *FAKE* coordinates: latitude 0.00000, longitude 0.00000, altitude 0 m
##### END #####

JSON up: {"stat":{"time":"2022-08-11 17:30:29 GMT","lati":0.00000,"long":0.00000,"alti":0,"rxnb":0,"rxok":0,"rxfw":0,"ackr":0.0,"dwnb":0,"txnb":0,"temp":30.0}}
```

closed 2022-08-18 by KevinWassermann94