NebraLtd/hm-pktfwd #108: 2022.08.17.6 (083026e) breaks radio

ifeign, 2022-10-28:


shawaj, 2022-10-28:
@Crash0v3r1de thanks for the ping. Have moved the issue to the correct repo.

Ping @KevinWassermann94 @MuratUrsavas

shawaj, 2022-10-28:
Not 100% sure this is the cause but...`"gps_i2c_path": "/dev/i2c-1",` was added back into the config files and I don't know why

Crash0v3r1de, 2022-10-28:
> Not 100% sure this is the cause but...`"gps_i2c_path": "/dev/i2c-1",` was added back into the config files for sx1302 and I don't know why.
> 
> My guess is this isn't a rak specific issue but is actually happening for all sx1302/sx1303
> 
> If I remember correctly we removed the GPS stuff from the concentrator code as it isn't needed, but it's been a long time since I've looked at it TBH so might be wrong
> 
> If it's not this GPS thing my next guess would be that something in the configuration files is going beyond the limits of the sx1250 outputs or something.

Thanks for migrating to the main repo. Not sure if logs and config files would help with this but I can send over whatever you'd want to look at.

shawaj, 2022-10-28:
If you do have any logs from a unit with this issue, particularly from the packet forwarder container, that could be really helpful yes

shawaj, 2022-10-31:
@ifeign @Crash0v3r1de @arduino43 @uros76 

I managed to reproduce this issue, but the issue is actually that you are using a testnet version of the software that was never meant for production. 

2022.08.17.6 is the latest production firmware.

2022.08.17.6-2 was never publicly released and had mangled config files. It only went to testnet.

2022.08.17.6-3 is the latest test version and works. But it hasn't been released to production yet either.

And the last update we pushed to the helium-rak fleet was 9th September. You should only use the officially released one at https://hub.balena.io/organizations/nebraltd/fleets/helium-rak and onboard your device by clicking `Get Started` not on `Fork this fleet`

Perhaps you have forked the fleet and are tracking a non-production version somewhere?

You can also see the latest official version at https://hub.balena.io/organizations/nebraltd/fleets/helium-rak/releases

Closing this issue as it's not an issue on our side.

closed 2022-10-31 by shawaj

shawaj, 2022-10-31:
For your reference, the latest testnet compose files are always stored here on the master branch of the main repo - https://github.com/NebraLtd/helium-miner-software/tree/master/device-compose-files

The latest production ready compose files are on the production branch - https://github.com/NebraLtd/helium-miner-software/blob/production/device-compose-files

The helium-rak repo config file (https://github.com/NebraLtd/helium-rak) should always mirror the testnet version shown above, but not always.

As I said, best way is to just use the fleet as it is supposed to be used (using Get Started button) not forking fleet. Failing that, you should use the production device-compose-files from the above link

shawaj, 2022-10-31:
@MuratUrsavas @KevinWassermann94 - no actual issue here just FYI ^^

Crash0v3r1de, 2022-11-01:
> @MuratUrsavas @KevinWassermann94 - no actual issue here just FYI ^^

Haven't had time to jump to the problem commit to grab logs but my hotspot did auto update to your latest commit and the radio status looks good now.
[image]

shawaj, 2022-11-01:
@Crash0v3r1de presumably you have forked the fleet and are updating it with your own workflows?

As if you were on the Balena hub fleet it never would have broken and wouldn't have just updated

Crash0v3r1de, 2022-11-01:
> @Crash0v3r1de presumably you have forked the fleet and are updating it with your own workflows?
> 
> As if you were on the Balena hub fleet it never would have broken and wouldn't have just updated

the only way for me to have control of my hotspot unfortunately.

shawaj, 2022-11-01:
I'm thinking that since the repo https://github.com/NebraLtd/helium-rak is meant for production devices I might change the action to pull in the production docker compose instead of the testnet one.

And perhaps I'll also create a testnet branch there which has the latest testnet one too?

What do you think @Crash0v3r1de ?

Crash0v3r1de, 2022-11-01:
> I'm thinking that since the repo https://github.com/NebraLtd/helium-rak is meant for production devices I might change the action to pull in the production docker compose instead of the testnet [...]

That would be super helpful for the folks who want the balena control over their devices.

shawaj, 2022-11-01:
Ok will sort this out this week sometime.

I think that makes most sense and will avoid similar issues occurring in the future 👌

Crash0v3r1de, 2022-11-01:
> Ok sorted that now @Crash0v3r1de - should be more stable now.
> 
> Master branch has the production config - https://github.com/NebraLtd/helium-rak/tree/master
> 
> But the testnet branch has the latest testing version - https://github.com/NebraLtd/helium-rak/tree/testnet
> 
> Haven't added any documentation about it but will do that soon

You are the best thank you!!!

shawaj, 2022-11-01:
No problem at all. Thanks to you and the others for bringing this to our attention 🙂