NebraLtd/hm-pktfwd #123: Setenv does not work properly with non-nebra fleets

shawaj, 2023-02-05:
Originally reported by @hellrc5 here https://github.com/NebraLtd/helium-rak/issues/19#issuecomment-1418269534

```
packet-forwarder Error: Cannot set FREQ variable.
packet-forwarder VARIANT variable is already set as COMP-RAKHM.
packet-forwarder Error: Not all required parameters have been set. Container will be reset after 5 sec.

Interestingly the diagnostics container detects the correct FREQ plan:

diagnostics INFO:hw_diag.utilities.hardware:Found the following LTE Devices: []
diagnostics INFO:root:Frequency: EU868
```

shawaj, 2023-02-05:
Relates-to: https://github.com/NebraLtd/hm-diag/issues/444
Relates-to: https://github.com/NebraLtd/hm-diag/issues/468

MuratUrsavas, 2023-02-06:
The best way to fix this, is to make FREQ parameter optional, not mandatory.

shawaj, 2023-02-06:
Closed by #126

closed 2023-02-06 by shawaj