NebraLtd/hm-pktfwd #140: packet-forwarder unable to start if interface wlan0 is not present

ctlynx, 2023-04-18:
**Problem:**
if `/sys/class/net/wlan0/address` does not exist, then pktfwd fails and is unable to start.

**Error:**
```
Traceback (most recent call last):
  File "/opt/pktfwd-dependencies/hm_pyhelper/miner_param.py", line 323, in get_ethernet_addresses
    diagnostics[key] = get_mac_address(path)
                       ^^^^^^^^^^^^^^^^^^^^^
  File "/opt/pktfwd-dependencies/hm_pyhelper/miner_param.py", line 350, in get_mac_address
    raise MinerFailedToFetchMacAddress("Failed to find file"
  File "/opt/pktfwd-dependencies/hm_pyhelper/miner_param.py", line 344, in get_mac_address
    file = open(path)
           ^^^^^^^^^^
hm_pyhelper.exceptions.MinerFailedToFetchMacAddress: Failed to find filecontaining miner mac address. Exception: [Errno 2] No such file or directory: '/sys/class/net/wlan0/address'
```
Log file: [nebra-error-log.txt](https://github.com/NebraLtd/hm-pktfwd/files/11265788/nebra-error-log.txt)
 
**Cause:**
Method [`get_ethernet_addresses`](https://github.com/NebraLtd/hm-pktfwd/blame/master/pktfwd/utils.py#L123) raises `MinerFailedToFetchMacAddress` and [`populate_local_conf_template`](https://github.com/NebraLtd/hm-pktfwd/blame/master/pktfwd/utils.py#L121) fails even if the MAC of `eth0` could be obtained.

I suppose that [`get_ethernet_addresses`](https://github.com/NebraLtd/hm-pktfwd/blame/master/pktfwd/utils.py#L123) should not fail and simply return the MAC addresses that do exist.
 
**Expected behavior:**
Miner should be able to startup and operate using `eth0` MAC address as before.

It may happen that `wlan0` does not exist if wifi dongle is broken or unplugged. 
On the device I detected the problem, `wlan0` does not exist because the dongle is broken. I have not replaced it with a new one because it does not use wifi. So far it was working well connected via ethernet. The dashboard showed that there was no MAC address for wifi, but it worked ok.

HelPi7, 2023-04-19:
Having the same issue!

shawaj, 2023-04-22:
Thanks @HelPi7 @ctlynx we're working on a fix.

Also related to https://github.com/NebraLtd/hm-diag/issues/557

robputt, 2023-04-24:
See https://nebraltd.atlassian.net/browse/NEBD-181

PR #141 (2023-04-24): fix: bump hm-pyhelper
bump hm-pyhelper to v0.14.9

Closes: #140

**Issue**

- Link:
- Summary:

**How**

**Screenshots**

**References**

**Checklist**

- [ ] Tests added
- [ ] Cleaned up commit history (rebase!)
- [ ] Documentation added
- [ ] Thought about variable and method names
commit 7eec8b9e (2023-04-24): fix: bump hm-pyhelper

bump hm-pyhelper to v0.14.9

Closes: #140
closed 2023-04-24 by robputt